lemariva/uPyLoRaWAN #6: Is it compatible with any LoRaWAN gateway?

sasilva1998, 2020-04-01:
I have been working with Heltec HTM01, using its ESP32 LoRa with arduino, but actually I would prefer to do it with uPy, is there a way I could migrate my project so that I can use this modules with HTM01? 

Thanks in advance, it is a great work, tried doing something similar [FunPython uPyLoRa](https://github.com/FunPythonEC/uPyLoRa), so that I could have an easier use of Weib modules, but lost track of it.

lemariva, 2020-05-17:
Yes, it is compatible with every LoRaWAN gateway. 
Your device has an SX1276, therefore, you need to change pins in `device_config` in the [config.py ](https://github.com/lemariva/uPyLoRaWAN/blob/eebd391ba822190695efe3c6308bef43d711fdeb/config.sample.py#L41) file to match your board and the board should connect to your gateway.

sasilva1998, 2020-05-17:
thank you! @lemariva

closed 2020-05-17 by sasilva1998