lemariva/uPyLoRaWAN #16: receiving with interrupt callback not working

dukeduck1984, 2021-01-09:
Hi,

I have set the `dio_0` pin number in `config.py` and changed the code in `examples/LoRaReceiver.py` as following:

```
def receive(lora):
    print('LoRa Receiver Callback')
    
    def cb(payload):
        lora.blink_led()
        print(payload)
    
    lora.on_receive(cb)
    lora.receive()
```

But it didn't work - couldn't receive any data.  It only worked with `while True` loop as shown in the original example.

Did I miss something?  Pls help!

It's an AI-Thinker Ra-02 LoRa module with ESP32.  The MicroPython firmware is V1.13 stable.

I used the sx127x driver in the master branch, and the readme title is uPyLora.

dukeduck1984, 2021-01-09:
I think I have found the reason - it's in the method `handle_on_receive`.

When the IRQ gets triggered, it's not necessarily that the full message has been received, hence `if (irq_flags == IRQ_RX_DONE_MASK)` may be `False` which then automatically set the module to single RX mode.  As a result, the callback funciton will never be called again.

It should be at least set a timeout loop to wait until the full message has come in and `irq_flags == IRQ_RX_DONE_MASK` for the callback to be executed.

I don't understand why in the original code it will in some cases set the module back to single RX mode - if the user has decided to use the module in countinuous mode, it should stick to that mode.

My modified asyncio version as following:

```
    async def on_receive(self, callback):
        self._on_receive_cb = callback
        if self._pin_rx_done:
            if callback:
                # enable RxDone interrupt in DIO Mapping 1
                self.write_register(REG_DIO_MAPPING_1, 0x00)
                self._pin_rx_done.irq(handler=self._pin_rx_done_handler, trigger=Pin.IRQ_RISING)
                await self.handle_on_receive()
            else:
                self._pin_rx_done.irq(handler=None, trigger=0)

    def _pin_rx_done_handler(self, pin):
        self._event_rx.set()

    async def handle_on_receive(self):
        while True:
            print('waiting rx')
            await self._event_rx.wait()
            print('irq triggered')
            rx_income_ms = utime.ticks_ms()
            self.set_lock(True)  # lock until TX_Done
            while utime.ticks_diff(utime.ticks_ms(), rx_income_ms) < 3000:  # Set timeout=3000ms
                irq_flags = self.get_irq_flags()
                if irq_flags == IRQ_RX_DONE_MASK:  # RX_DONE only, irq_flags should be 0x40
                    # automatically standby when RX_DONE
                    if self._on_receive_cb:
                        print('msg coming in')
                        payload = self.read_payload()
                        self._on_receive_cb(payload)
                        break
                else:
                    await asyncio.sleep(0)

            # elif self.read_register(REG_OP_MODE) != (MODE_LONG_RANGE_MODE | MODE_RX_SINGLE):
            #     print('no msg received')
            #     # no packet received.
            #     # reset FIFO address / # enter single RX mode
            #     self.write_register(REG_FIFO_ADDR_PTR, FifoRxBaseAddr)
            #     self.write_register(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_RX_SINGLE)
            self.set_lock(False)  # unlock in any case.
            self.collect_garbage()
```

jbfuzier, 2021-01-10:
Hi,

You are right, the irq can be called when there is an invalid crc or when the fifo is full.

From what I am understanding, the current lib does not handle cases for which the packet size is bigger than the fifo (64bytes).

What payload did you send to run into the issue ?

dukeduck1984, 2021-01-11:
Would you pls be more specific?  Did you mean the timing of IRQ being triggered versus the RX status?  Regarding the Regdiomapping configuration, I think it's set in the method `on_receive()` as following:

```
    def on_receive(self, callback):
        self._on_receive = callback

        if self._pin_rx_done:
            if callback:
                self.write_register(REG_DIO_MAPPING_1, 0x00)
                self._pin_rx_done.irq(
                    trigger=Pin.IRQ_RISING, handler = self.handle_on_receive
                )
            else:
                self._pin_rx_done.detach_irq()
```

According to the datasheet, writing `0x00` to `REG_DIO_MAPPING` enables the interrupt on `DIO0` pin upon RX.  Is there any other configurations need to be done?

jbfuzier, 2021-01-11:
Sorry, you are totally right. I read the datasheet to quickly and mixed up the packet mode (which can involves more complex cases to handle due to the FIFO being only 64B) with the LORA mode.

[image]
[image]
[image]

What value do you have when you read the irq flags register ?
Does it happen every time or only some times ? How long is the payload you are sending ?

dukeduck1984, 2021-01-11:
It happened every time - I never managed to receive messages until I modified the code as shown above.  The payload was 15 bytes long.

I didn't print out the irq flags value, so I have no idea what it was.  But according to the info you posted, the interrupt triggers when the payload is readily available (RxDone), no more waiting is needed.

dukeduck1984, 2021-01-11:
I read the datasheet again - the interrupt at DIO0 pin can occur in both Rx Single mode & Rx Continuous mode.  In Rx Single mode, a Rx Timeout period should be set first, the LoRa module will wait for the preamble for that Rx Timeout period of time before triggers RxTimeout interruption and goes to standby mode.  If the payload comes in before timeout, RxDone interruption will occur.

But in the original code, I didn't see anywhere that Rx Timeout period can be set (RegSymbTimeout).

dukeduck1984, 2021-01-11:
There's a bit difference in the code [here](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/modules_examples/lora/sx127x.py), in the comment it says the IRQ Flag should be 0x50, rather than 0x40 in this repo.

dukeduck1984, 2021-01-12:
So I printed the value of IRQ flag, it's 80 which was 0x50, or 0b01010000, which meant RxDone & ValidHeader per datasheet on page 111.  The working code as following:

```
    async def handle_on_receive(self):
        while True:
            print('waiting rx')
            await self._event_rx.wait()
            print('irq triggered')
            self.set_lock(True)  # lock until TX_Done
            irq_flags = self.get_irq_flags()
            # print(irq_flags)
            # irq_flags should be 0x50 (0b01010000): RxDone & ValidHeader
            if irq_flags & IRQ_RX_DONE_MASK == IRQ_RX_DONE_MASK:
                # automatically standby when RX_DONE
                if self._on_receive_cb:
                    print('msg coming in')
                    payload = self.read_payload()
                    self._on_receive_cb(payload)
            # elif self.read_register(REG_OP_MODE) != (MODE_LONG_RANGE_MODE | MODE_RX_SINGLE):
            #     print('no msg received')
            #     # no packet received.
            #     # reset FIFO address / # enter single RX mode
            #     self.write_register(REG_FIFO_ADDR_PTR, FifoRxBaseAddr)
            #     self.write_register(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_RX_SINGLE)
            self.set_lock(False)  # unlock in any case.
            self.collect_garbage()
```

jbfuzier, 2021-01-13:
> It happened every time - I never managed to receive messages until I modified the code as shown above. The payload was 15 bytes long.
>
> I didn't print out the irq flags value, so I have no idea what [...]

Strange, it worked for me on a TTGO Lora board V1. 

I agree with you regarding the rx single part in the code, I do not understand why it is here...

0x50 seems better but 0x40 should works fine too, worst case you get rubish data.

I spent a lot of time strugling with the IQ config as the datasheed is not very clear regarding this part. (LORAWan is using reverse IQ so that gateway can not see packets from other gateways; same thing for nodes which can not see packet from each other).

Would you share your asyncio version ? (I am very new to asyncio and I would be interested to learn it)

dukeduck1984, 2021-01-13:
0x40 only worked for me as shown in my 2nd post from the top.

I only rewrote the interrupt part with asyncio in order to have it working with my other asyncio code base, as the interrupt may cause a racing condition.  Below is the full driver for your reference (letter case also changed to comply with PEP)

```
import gc
import utime

from machine import Pin
from micropython import const

from primitives.irq_event import IRQ_EVENT

PA_OUTPUT_RFO_PIN = const(0)
PA_OUTPUT_PA_BOOST_PIN = const(1)

# registers
REG_FIFO = const(0x00)
REG_OP_MODE = const(0x01)
REG_FRF_MSB = const(0x06)
REG_FRF_MID = const(0x07)
REG_FRF_LSB = const(0x08)
REG_PA_CONFIG = const(0x09)
REG_LNA = const(0x0C)
REG_FIFO_ADDR_PTR = const(0x0D)

REG_FIFO_TX_BASE_ADDR = const(0x0E)
FifoTxBaseAddr = const(0x00)
# FifoTxBaseAddr = 0x80

REG_FIFO_RX_BASE_ADDR = const(0x0F)
FifoRxBaseAddr = const(0x00)
REG_FIFO_RX_CURRENT_ADDR = const(0x10)
REG_IRQ_FLAGS_MASK = const(0x11)
REG_IRQ_FLAGS = const(0x12)
REG_RX_NB_BYTES = const(0x13)
REG_PKT_RSSI_VALUE = const(0x1A)
REG_PKT_SNR_VALUE = const(0x1B)
REG_MODEM_CONFIG_1 = const(0x1D)
REG_MODEM_CONFIG_2 = const(0x1E)
REG_PREAMBLE_MSB = const(0x20)
REG_PREAMBLE_LSB = const(0x21)
REG_PAYLOAD_LENGTH = const(0x22)
REG_FIFO_RX_BYTE_ADDR = const(0x25)
REG_MODEM_CONFIG_3 = const(0x26)
REG_RSSI_WIDEBAND = const(0x2C)
REG_DETECTION_OPTIMIZE = const(0x31)
REG_DETECTION_THRESHOLD = const(0x37)
REG_SYNC_WORD = const(0x39)
REG_DIO_MAPPING_1 = const(0x40)
REG_VERSION = const(0x42)

# invert IQ
REG_INVERTIQ = const(0x33)
RFLR_INVERTIQ_RX_MASK = const(0xBF)
RFLR_INVERTIQ_RX_OFF = const(0x00)
RFLR_INVERTIQ_RX_ON = const(0x40)
RFLR_INVERTIQ_TX_MASK = const(0xFE)
RFLR_INVERTIQ_TX_OFF = const(0x01)
RFLR_INVERTIQ_TX_ON = const(0x00)

REG_INVERTIQ2 = const(0x3B)
RFLR_INVERTIQ2_ON = const(0x19)
RFLR_INVERTIQ2_OFF = const(0x1D)

# modes
MODE_LONG_RANGE_MODE = const(0x80)  # bit 7: 1 => LoRa mode
MODE_SLEEP = const(0x00)
MODE_STDBY = const(0x01)
MODE_TX = const(0x03)
[... 422 lines cut ...]
    def transfer(self, address, value=0x00):
        response = bytearray(1)
        self._pin_ss.value(0)
        self._spi.write(bytes([address]))
        self._spi.write_readinto(bytes([value]), response)
        self._pin_ss.value(1)
        return response

    def reset_module(self, duration_low_ms=50, duration_high_ms=50):
        self._pin_rst.off()
        utime.sleep_ms(duration_low_ms)
        self._pin_rst.on()
        utime.sleep_ms(duration_high_ms)

    def set_pins(self, ss, rst, dio0=None):
        self._pin_ss = Pin(ss, Pin.OUT, value=1)
        self._pin_rst = Pin(rst, Pin.OUT, value=1)
        self._pin_rx_done = Pin(dio0, Pin.IN) if dio0 else None

    @staticmethod
    def collect_garbage():
        gc.collect()
        # print('[Memory - free: {}   allocated: {}]'.format(gc.mem_free(), gc.mem_alloc()))

```

dukeduck1984, 2021-01-13:
@jbfuzier Is the invert IQ documented somewhere?  I'd like to learn about it in case someday I will be using LoRaWan as well.  Thanks.

jbfuzier, 2021-01-13:
Thanks.

Unfortunately I have not found much information regarding this invert IQ mecanism.

Ultimately, the goal is to reduce useless load on the node & gateway by making sure nodes don't see messages from each other and gateway does not see messages from each other (which would be useless). 

To do that Gateways send messages with IQ inverted (which physically changes the phase of the signal), nodes must be expected IQ to be inverted in order to receive the messages.

The datasheet is very misleading (not to say wrong) regarding the INVERTIQ register

[image]

For example if you want to only have nodes that can all talk to each other (= do not invertIQ), you need to set the register to 0x01, otherwise nodes can not talk to each other.

If you want to have a gateway/nodes scenario, register value need to be 0x41 on gateway and 0x00 on the nodes (or the other way around).

To add lorawan I think the best course of action is to port a working arduino lib or a python lib to micropython. I am not using Lorawan (too complex for a private network IMHO), I just have a use case with several nodes and one receiver; I added on top a simple layer which do the encryption, integrity check and ack/retry. The issue is that encryption adds a lot of overhead to the payload 32bytes for AES128.

dukeduck1984, 2021-01-13:
Thanks for the explaination.  My use case is similar to yours - multiple nodes & one receiver, TTN not involved.  What hardware (LoRa module & microcontroller) do you use for the receiver?  Same as the nodes?

saramonteiro, 2021-05-04:
Hi @dukeduck1984 @jbfuzier I am using a customized board and I also faced the same issue.
The RX handler is not being triggered at all. Only the original polling mode is working for me.
I even used a print inside the `handle_on_receive` to see if the interruption was being triggered,
but it seems it isn't. 
Did it also happen to you @dukeduck1984 ?
I mean, was your handler triggered and you couldn't retrieve the data or have your callback called or your handler was
not triggered at all, just like me? 
I still didn't have the chance to try your async mode, I have to understand how it works.

dukeduck1984, 2021-05-04:
You need to change the `if` statement inside the `handle_on_receive` to: `if irq_flags & IRQ_RX_DONE_MASK == IRQ_RX_DONE_MASK:`, then it will work.

saramonteiro, 2021-05-04:
Hum.. ok, thank you @dukeduck1984 
But in my case,  `handle_on_receive` isn't being triggered.

saramonteiro, 2021-05-05:
@dukeduck1984  I used a LA to monitor the DI0 pin and I figured out it is not changing level and I have another node continuously sending data.

saramonteiro, 2021-05-06:
@dukeduck1984 my issue was solved.
I forgot to call lora.receive() to change the LoRa mode. 
¬¬

Bujtar, 2021-09-28:
@dukeduck1984 
The IRQ_EVENT is deprecated in the newer version of uasyncio. (https://forum.micropython.org/viewtopic.php?t=9867)
Do you have maybe an updated your lorawan program with this new version?
If not, could please guide me, how to do that?

Thanks is advance.

dukeduck1984, 2021-09-28:
Hi @Bujtar ,

As Peter pointed out in his post, you can use `ThreadSafeFlag`, as per the example shown [here](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#36-threadsafeflag).

kjm1102, 2023-05-15:
According to note [1] at the bottom of https://lora-developers.semtech.com/documentation/tech-papers-and-guides/channel-activity-detection-ensuring-your-lora-packets-are-sent/how-to-ensure-your-lora-packets-are-sent-properly/ the sx1276 should be able to receive lora packets without host mcu involvement.

I'm trying use dio0 to wake an esp32 from deepsleep when the sx1276 receives a packet but since the dio0 pin never moves it's not working.  I think maybe I need to configure the sx1276 for continuous receive to achieve this?

davefes, 2023-05-16:
Would you tell me how you configured DIO0 to wake-up the ESP32?  What happens after the ESP32 wakes-up? Can you tell where things stall?

kjm1102, 2023-05-16:
Let me put what I did in a proper program that you can easily run instead of the interactive nightmare I was developing with. That way I can also send you a copy of the while loop code benchmark code I was using previously with the much reduced sx1276 current. Might take me a day or so to put some lipstick on the pig. I'll also put it on https://github.com/orgs/micropython/discussions/11515 so our lora chats are all in one place.

davefes, 2023-05-17:
Good. I will take me some time to swap-over to his code on my test link.