lemariva/uPyLoRaWAN #21: Invalid Version

ReadieS, 2024-04-17:
Hi,
 I keep getting an error when I try and use this code. sometimes it works flawlessly and rather than the error I get "SX Verrsion 18" but 50% of the time i get the following 

`MPY: soft reboot
error with BME280 Module
Found DS devices:  []
error with DS18 Module
Traceback (most recent call last):
  File "<stdin>", line 31, in <module>
  File "sx127x.py", line 121, in __init__
Exception: Invalid version.`

my code begins with:
`import machine
import time
import math
try:
    import BME280_Read
except:
    print("error with BME280 Module")
try:
    import DS18_Read
except:
    print("error with DS18 Module")
import wind_direction
import statistics
import datetime
import _thread
import struct
import ujson

#lora specific setup for upylora
from config import *
from machine import Pin, SoftSPI
from sx127x import SX127x
import sender

device_spi = SoftSPI(baudrate = 10000000, 
        polarity = 0, phase = 0, bits = 8, firstbit = SoftSPI.MSB,
        sck = Pin(device_config['sck'], Pin.OUT, Pin.PULL_DOWN),
        mosi = Pin(device_config['mosi'], Pin.OUT, Pin.PULL_UP),
        miso = Pin(device_config['miso'], Pin.IN, Pin.PULL_UP))

lora = SX127x(device_spi, pins=device_config, parameters=lora_parameters)
#end of lora specific setup
...............
`

I'm using a LILYGO TTGo T3 V1.6 board 

Here is the snippet when it does work
`MPY: soft reboot
error with BME280 Module
Found DS devices:  []
error with DS18 Module
SX version: 18
Measuring wind direction for 5 seconds...`

ReadieS, 2024-04-20:
I think I have sorted it. in sx127x.py I have substituted
`# check hardware version
        init_try = True
        re_try = 0
        while init_try and re_try < 10:
            version = self.read_register(REG_VERSION)
            re_try = re_try + 1
            if version != 0:
                init_try = False
        if version != 0x12:
            raise Exception('Invalid version.')

        if __DEBUG__:
            print("SX version: {}".format(version))

`

For the folowing 

`        # Number of consecutive matching versions required
        consecutive_matches = 2

        # Initialize variables to track consecutive matches
        matched_versions = 0
        last_version = None

        # check hardware version
        while matched_versions < consecutive_matches:
            init_try = True
            re_try = 0
            while init_try and re_try < 10:
                version = self.read_register(REG_VERSION)
                re_try = re_try + 1
                if version != 0:
                    init_try = False
                else:
                    time.sleep(0.002)  # Pause for 2 milliseconds between retries

            if version != last_version:
                last_version = version
                matched_versions = 1
            else:
                matched_versions += 1

        if version != 0x12:
            raise Exception('Invalid version.')

        if __DEBUG__:
            print("SX version: {}".format(version))
`

also had to add  ` import time` at the top

volnix, 2025-03-04:
I'm getting the same issue.  Board is a [LilyGo LoRa32 v2.1](https://lilygo.cc/products/lora3?srsltid=AfmBOorX7UZ5bdkG7MPJB9IqPeqtcWbRGc7AramM5WeU9mZ9qo65JpWE).  I tried the code change above, but still no luck.  The only way to fix is to re-flash micropython then re-install code, then it works perfectly.  Subsequent reads to the register yield `0x00` no matter what.

volnix, 2025-03-04:
Actually, all registers read `0x00` it turns out.  This points more to a bug in Micropython's SPI library for this board (???)

volnix, 2025-03-05:
Fixed this!  I had to re-init the SPI in the program initiatilization phase.  e.g., 

```
device_spi.deinit()
device_spi.init(baudrate = 10000000, 
        polarity = 0, phase = 0, bits = 8, firstbit = SPI.MSB,
        sck = Pin(device_config['sck'], Pin.OUT, Pin.PULL_DOWN),
        mosi = Pin(device_config['mosi'], Pin.OUT, Pin.PULL_UP),
        miso = Pin(device_config['miso'], Pin.IN, Pin.PULL_UP))
```